stereolabs/zed-docker #4: Positional tracking sample issues

orshefi, 2019-02-13:
Hey guys,

I'm trying to use your docker container with the Zed SDK 2.7.1 and UBUNTU 16.04 and CUDA 10.0(With X server support and OpenGL).

Setup and everything went pretty well, here are the tests I made and the issues I encountered:
* I have compiled the positional tracking sample and ran it (Both on my machine and inside the docker container - same hardware).
* In both cases, the app ran out of the box with no errors/exceptions.
* There was a big difference in performance:
** The version I ran on my machine ran without any position tracking losses (on few iterations).
** The version I ran on the docker container right from the start had a lot of tracking losses.
** I had also recorded an svo file and ran positional tracking on it, there was **no loss** at all.
* I ran every test with "nvidia-docker" and "docker --runtime nvidia" that are FFF by definition but only to make sure 😊.

These tests led me to think that there were issues with the camera streams but it seems on ZED Explorer that there is no glitches visable.

I had checked it with all the docker images offered by your github repo as well, all of those underperformed.

Please let me know if I am missing out something.

adujardin, 2019-02-14:
Hi,
If I understand correctly there is a performance hit when using the ZED in USB (not SVO file) and/or possibly using the graphical interface in docker?

> * ** I had also recorded an svo file and ran positional tracking on it, there was **no loss** at all.

Does this mean you tested an SVO file in docker with GUI and it worked as well as natively?

To be honest we didn't thoroughly bench the performances using Docker, but the fact that it works reasonably well but not completely is quite puzzling :confused: 

To check if the camera stream is identical in docker you can run the Diagnostic app (`/usr/local/zed/tools/ZED\ Diagnostic` ) within a Docker image to check for frame drops or any image issues.
You could also display the depth to visually check any potential obvious defect using `ZED Depth Viewer`.

Also, do your CPU and GPU usage appears to be similar when running the same sample in docker and natively? I'm thinking the extra load in docker could drop some frame in live mode.
You could actually test this by running the sample with a SVO in [real-time mode](https://www.stereolabs.com/docs/api/structsl_1_1InitParameters.html#af0759d9c02cf9cebb5e385637173fd4a). For this you have to set the `InitParameters.svo_real_time_mode` to `true`.

orshefi, 2019-02-17:
I don't know what changed but the issue went away.

Also, I did run ZED Diagnostics, it runs with flying colors but alerts me on lens smudge.
You can imagine that I relentlessly tried to clean the lenses and nothing seems to solve it.
You should also know that this issue won't appear on the same container outside the docker container.

Any Ideas?

closed 2022-04-27 by github-actions[bot]