stereolabs/zed-docker #6: Passing ZED devices to docker container

orshefi, 2019-03-17:
Hi,

I want to pass zed USB devices and video devices to the container.

The reason is that I want to use docker container to each camera I connect to the PC (I have several of those)

For now I passed:
- /dev/video0
- /dev/video1
- /dev/bus/usb/<bus_num>/<dev_num> - hid device
- /dev/bus/usb/<bus_num>/<dev_num> - uvc device

For a single camera, this setup works fine
When I connect dual camera setup (and pass each of the containers the proper video and USB devices) none of the containers works

What am I missing?

adujardin, 2019-03-18:
Hi,
It seems to be related to the way each USB device is handled by docker, and to be honest we didn't thoroughly test the multi-camera setup.

However, when trying with the `--privileged` I successfully launch 2 containers with 2 ZED plugged in. Both containers are able to see and open the 2 ZED. I understand this isn't ideal since they have all access to everything and it therefore requires a mechanism to open the correct one.

Unfortunately, I don't have a better solution for now.

orshefi, 2019-03-22:
I do have a mechanisem that query the serial number of the cameras and initialize the right camera (compared to a config file) but it consists of pseudo random behavior, sometimes one of the camera is opened, sometimes both.

Can you work with us to reach a solution? docker is a must for our application

adujardin, 2019-03-22:
1. So to understand the problem, you need exactly 1 camera detected for each docker container and you know the bus_num/dev_num in advance, right?

2. Is it an option to connect all the camera to every docker container and only open the right one, provided that it detect and open the right one reliably?

3. Did you try to use the SDK function [Camera::getDeviceList()](https://www.stereolabs.com/docs/api/classsl_1_1Camera.html#a1768d383f5e78746f58eda9d2c60b33e) to select the correct camera?

4. Do you work with ZED-M (which also have IMU as separated USB device) ?

5. Could you provide a command or a docker-compose sample to try to reproduce the error with the same configuration (single/dual camera setup), as mentioned in your first post?

closed 2020-02-28 by adujardin