stereolabs/zed-docker #14: Jetpack 4.2 python-runtime seg fault

rbrigden, 2020-03-12:
We start a streaming sender using the python example in `stereolabs/zed-examples` in the `stereolabs/zed:3.1-py-runtime-jetson-jp4.3` and encounter a segfault on our Jetson Nano.

More specifically, we build an image on the Nano with this docker file

```
FROM stereolabs/zed:3.1-py-runtime-jetson-jp4.3
COPY /main.py /main.py
CMD ["python3", "/main.py"]
```

where `main.py` is an exact copy of [this](https://github.com/stereolabs/zed-examples/blob/master/camera%20streaming/sender/python/streaming_sender.py)

We build and run the image as so

```
docker build . -t zed-sender
docker run --rm zed-sender
Segmentation fault (core dumped)
```

adujardin, 2020-03-13:
I am able to reproduce it. It seems there's a bug loading the Python API in docker on Nano.
We'll investigate and keep you informed

adujardin, 2020-03-16:
So trying different things, I think it boils down to the docker image launch options, without `--gpus all` on jetson the sample segfault (and without `--privileged` no camera will be detected).
Not sure yet why it doesn't output a cleaner error code. For now, you should try with :
```
docker run --gpus all --privileged --rm zed-sender
```

rbrigden, 2020-03-17:
Thanks! Forgot to add that flag...

This solved the sender crash issue, but we still were not able to read the stream from the receiver. I re-flashed the device with Jetpack 4.3 and found this command to work perfectly. Using nvidia-docker2 here

```
sudo docker run -it --rm --runtime nvidia -v streaming_sender.py:/main.py --privileged --network=host stereolabs/zed:3.1-py-runtime-jetson-jp4.3  python3 /main.py
```

`--network=host ` was the next trick to make it work.

adujardin, 2020-03-17:
Yes, by default Docker isolate as much as possible, I'll add the clarification in the readme about the network.

closed 2020-03-17 by adujardin