stereolabs/zed-docker #25: Segmentation fault on Jetson Nano when running ZED_Explorer.

NicolajFoerderer, 2020-08-05:
Hi,
I am running a container on a NVIDIA Jetson Nano with display support:
```bash
xhost +si:localuser:root
sudo docker run -it --gpus all --privileged -e DISPLAY=$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix stereolabs/zed:3.2-tools-devel-jetson-jp4.3
```

When I run `tools/ZED_Diagnostic` all tests pass, but when I run `tools/ZED_Explorer` the GUI pops up and then crashes with the following error message:

```bash
nvbuf_utils: Could not get EGL display connection
Segmentation fault (core dumped)
```

Same issue, with other versions of ZED SDK or JetPack. If I install the ZED SDK on the Jetson Nano directly, everything works fine.

leicmi, 2020-08-18:
Somewhat related, does using `--net=host` as described in <https://stackoverflow.com/questions/58851228/gstreamer-camera-usage-within-docker-containers> solve your issue?
For me it solves the error `nvbuf_utils: Could not get EGL display connection` in a very different setup.
I'm still looking for a way to avoid it though...

NicolajFoerderer, 2020-08-18:
That's it! Thank you dear stranger.

closed 2020-08-18 by NicolajFoerderer