stereolabs/zed-docker #26: Missing lnvcuvid.so.1

mmartin4972, 2020-09-19:
Hello, 

I've upgraded from ZED SDK 2.8 to 3.2 and am trying to get my refactored code to pass our docker build. I've followed the steps for downloading and installing the new version of the ZED SDK in the docker image, however when I run my test code I get the following error:

`/usr/bin/ld: warning: libnvcuvid.so.1, needed by /usr/local/zed/lib/libsl_zed.so, not found (try using -rpath or -rpath-link)`

I've tried to create an nvidia-docker to enable nvidia drivers in the docker image, however nvidia-docker requires an nvidia gpu which is not something that the Travis CI system supports. Do you have any ideas how I can solve this issue other than creating an nvidia-docker image? Any help would be greatly appreciated.

[Here](https://travis-ci.org/github/umrover/mrover-workspace/builds/724334364) is the link to my build.

obraun-sl, 2020-09-21:
Hi,
nvcuvid is installed by NVIDIA drivers and is required for the sdk. 
If you want to use travis for build only, I think it's possible to have a cuda compatible travis file, but that will work for build only.
Maybe the following repository can help you : 
https://github.com/jeremad/cuda-travis

mmartin4972, 2020-09-26:
That was very helpful. Thanks for you response!

closed 2020-09-26 by mmartin4972