stereolabs/zed-docker #28: Terminate called after throwing an instance of 'pwgen::PwgenException'   what():  Driver error:

phamdat09, 2020-11-20:
Hello,

I try to run the ZED-SDK 3.3.1 with cuda 11.1, ubuntu 18.04 and driver-nvidia version 455. ( I have a card RTX 2080 Ti).

When I run the `object dection` module with an optimization, i got an erorr `Terminate called after throwing an instance of 'pwgen::PwgenException'   what():  Driver error: `

Can you tell me How I can fix it ??
Thanks

adujardin, 2020-11-20:
Hi,

Could you specify the exact commands used to get this error?
This is not an error we encountered before. 

In the meantime, you could try to ask the Nvidia developers at https://forums.developer.nvidia.com/c/accelerated-computing/gpu-accelerated-libraries/12 if they have any idea on the possible root cause.

phamdat09, 2020-11-23:
Hello, i use this docker file to build an image!!

```
FROM nvidia/cuda:11.1-devel-ubuntu18.04
RUN apt-get update -y && apt-get install --no-install-recommends lsb-release wget less udev sudo build-essential cmake -y && \
    wget -qO ZED_SDK.run https://download.stereolabs.com/zedsdk/3.3/cu111/ubuntu18_full && \
    chmod +x ZED_SDK.run ; ./ZED_SDK.run silent && \
    rm ZED_SDK.run && \
    rm -rf /var/lib/apt/lists/*

ENV LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/cuda/lib64:/usr/local/cuda/lib64/stubs:/usr/lib/x86_64-linux-gnu
ENV CMAKE_LIBRARY_PATH=$CMAKE_LIBRARY_PATH:/usr/local/cuda/lib64:/usr/local/cuda/lib64/stubs:/usr/lib/aarch64-linux-gnu
WORKDIR /usr/local/zed/
```
After, i use this commands:
`sudo docker run --gpus all -e NVIDIA_DRIVER_CAPABILITIES=all -ti zed_devel:latest bash` with `zed_devel:latest` is the name of docker imgae.
Then, I will build the object detection with C++.It raises an error which I mentioned when the model optimizea!!

Thanks for your supports

adujardin, 2020-11-23:
OK, I have the same message and I found this issue. You shouldn't add the stubs folder to the ld library path. It's dummy libraries only meant for building.

Your Dockerfile should be like (notice the `LD_LIBRARY_PATH`): 

```
FROM nvidia/cuda:11.1-devel-ubuntu18.04
RUN apt-get update -y && apt-get install --no-install-recommends lsb-release wget less udev sudo build-essential cmake -y && \
    wget -qO ZED_SDK.run https://download.stereolabs.com/zedsdk/3.3/cu111/ubuntu18_full && \
    chmod +x ZED_SDK.run ; ./ZED_SDK.run silent && \
    rm ZED_SDK.run && \
    rm -rf /var/lib/apt/lists/*

ENV LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/cuda/lib64:/usr/lib/x86_64-linux-gnu
ENV CMAKE_LIBRARY_PATH=$CMAKE_LIBRARY_PATH:/usr/local/cuda/lib64:/usr/local/cuda/lib64/stubs:/usr/lib/aarch64-linux-gnu
WORKDIR /usr/local/zed/
```

To test this I used your original Dockerfile and removed the stubs folder (`rm -rf /usr/local/cuda/lib64/stubs/`) in interactive mode to make sure, and the optimization worked.

phamdat09, 2020-11-23:
Thank you so much!!

Your solution works well on my side !!

closed 2020-11-23 by phamdat09