stereolabs/zed-docker #31: Installing ZED SDK in docker

oscarthorn, 2020-12-10:
Hi!

I'm trying to install the ZED SDK in a docker container. Here is the relevant part of the dockerfile:

`RUN mkdir -p ~/Downloads && cd ~/Downloads && wget https://download.stereolabs.com/zedsdk/3.1/jp43/jetsons && chmod +x jetsons && ./jetsons`

This fails with `This script shouldn't be run as root, the root permission will be asked when needed (to install the dependencies, and the SDK into /usr/local/zed)` but in docker build there is only the root user. Should I do this in some other way. Could not really figure it out from you docker files.

Thanks!

adujardin, 2020-12-10:
You can run the installer like this `./installer.run -- silent`

The space between `--` and `silent` has its importance. It will run as in the Dockerfile we're using and will not check if it's root (and will also run as "_noninteractive_").

The alternative is to deflate the installer, then modifying the install script, like this:
```
./installer.run --noexec --target installer_files
cd installer_files
# Modify linux_install_release.sh
./linux_install_release.sh
```

closed 2021-12-14 by adujardin

smokemonkey101, 2024-10-04:
I cant get the sdk tools show up in the Docker sdk 
```
sudo apt-get update && sudo apt-get install -y nvidia-docker2 nvidia-container-toolkit
sudo systemctl daemon-reload
sudo systemctl restart docker

docker pull stereolabs/zed:4.1-devel-cuda12.1-ubuntu20.04
xhost +si:localuser:root 
docker run -it --runtime nvidia --privileged -e DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix stereolabs/zed:4.1-devel-cuda12.1-ubuntu20.04
```
In the bash i ran /usr/local/zed/tools/ZED_Explorer
I get  "bash: /usr/local/zed/tools/ZED_Explorer: No such file or directory"

[image]