stereolabs/zed-docker #35: ZED SDK cannot be installed on Jetson AGX with docker image Ubuntu 20.04 focal v8 ARM!!

Bmoradi93, 2021-06-21:
Hello stereolabs team. 
We are using your ZED2 cameras on all our robots.
We are trying to build a docker image and its base image is gonna be arm64v8/ubuntu: focal. 
The reason we are using this base image is that we are trying to use ZED2-ROS2 Wrapper.

Problem: ZED SDK cannot be installed on that docker image and the reason is CUDA 10.2 which cannot be installed on Ubuntu 20.04.

I appreciate it if you could provide a docker image as a solution to this issue.
Any other solutions will be appreciated!
Thank you

adujardin, 2021-06-22:
Hi,

On jetson, the CUDA libraries are mounted directly from the host and you can't easily use custom images, you have to use the l4t-base provided by Nvidia, matching the jetpack installed https://ngc.nvidia.com/catalog/containers/nvidia:l4t-base

There are images with ROS2 installed available for jetson however if needed https://github.com/dusty-nv/jetson-containers that can be used as base image

Bmoradi93, 2021-06-29:
Thank you for your response,
Is https://ngc.nvidia.com/catalog/containers/nvidia:l4t-base based on Ubuntu 20.04 ?
Because ROS2 foxy is available on for Ubuntu 20.04 OS.
I have installed Ubuntu20.04 LTS on my Jetson Xavier and I installed ROS-Foxy. ZED2_ROS2_Wrapper works fine on it.
The only thing we need is a docker image based on Ubuntu 20.04 and Cuda 10.2.

adujardin, 2021-06-30:
Yes, it's based on the same versions as the host, since it's a requirement for jetson. It supports JetPack 4.4 and 4.5, you should match the image tag (like `r32.4.4` or `r32.5.0`)

closed 2021-12-14 by adujardin