stereolabs/zed-docker #38: Missing libusb-1.0.so

mskrzypkows, 2021-12-14:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I've run container from image `stereolabs/zed:3.6-gl-devel-cuda11.4-ubuntu20.04`, software which built fine on `3.5-gl-devel-cuda11.1-ubuntu20.04`, doesn't build on 3.6 based image. CMake cannot find `libusb-1.0.so` after creating symbolic link:
```ln -s /lib/x86_64-linux-gnu/libusb-1.0.so.0.2.0 /usr/lib/x86_64-linux-gnu/libusb-1.0.so``` cmake finds the libusb without problem.

### Steps to Reproduce

1. `docker run --gpus all --rm -it -v /dev:/dev --net=host -e DISPLAY=:0 --privileged=true stereolabs/zed:3.6-gl-devel-cuda11.4-ubuntu20.04 bash`
2.  `cp -r /usr/local/zed/samples/depth\ sensing/ ~`
3. `cd ~/depth\ sensing/cpp/`
4. `mkdir build`
5. `cd build`
6. `cmake ..`
7. `make`
...

### Expected Result

Output binary created

### Actual Result

```** No rule to make target '/usr/lib/x86_64-linux-gnu/libusb-1.0.so', needed by 'ZED_Depth_Sensing'.  Stop.```

### ZED Camera model

ZED2i

### Environment

```shell
OS: Manjaro Linux
```

adujardin, 2021-12-14:
Thank you for reporting the issue, it should be fixed for the next patch release

adujardin, 2022-01-07:
It's now fixed (you should manually re-pulled the image to have the fix applied since the tag doesn't change on patch release)

closed 2022-01-07 by adujardin