stereolabs/zed-docker #39: Docker SVO recording fails using hardware encoding (H264/H265)

timongentzsch, 2022-02-01:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

When starting an SVO recording in a Docker container I get an `Segmentation fault (core dumped)` error. All other functions of the SDK (depth sensing, ground plane estimation, object detection, etc.) work without problems. Even recording in `LOSSLESS` compression mode works, so the problem is most likely due to the use of `NVENC`. 

Since I'm not using the official Jetpack image, but a yocto build with the [BSP layer for NVIDIA Jetson](https://github.com/OE4T/meta-tegra), it may well be that certain libraries are in different places. For example, I had to set up this symlink: `ln -s /usr/lib/aarch64-linux-gnu/libv4l/plugins/libv4l2_nvvidconv_wrapper.so /usr/lib/aarch64-linux-gnu/libv4l/plugins-wrapped/libv4l2_nvvidconv.so` beforehand so that the corresponding library could be found. However, since I do not receive any further information besides the SEGFAULT error message, I do not know exactly where the problem is. Is there a way to further debug this?

I already ran the `ZED_Diagnostic` tool and got following result:
<details>
  <summary>ZED Diagnotic File</summary>

``` json
{
    "Camera Test": {
        "ZEDCount": 1,
        "camera": {
            "Device ID": 0,
            "Firmware": 1523,
            "Init Output": "OK",
            "Model": "ZED 2",
            "Serial Number": 22557364,
            "valid": [
                "<b>Camera: </b>ZED 2",
                "<b>Firmware: </b>1523"
            ]
        },
        "resolutions": {
            "HD1080@15": {
                "fps (input)": 15,
                "fps (output)": 15,
                "height (input)": 1080,
                "height (output)": 1080,
                "initialization": "OK",
                "status": true,
                "width (input)": 1920,
                "width (output)": 1920
            },
            "HD1080@30": {
                "fps (input)": 30,
                "fps (output)": 30,
                "height (input)": 1080,
                "height (output)": 1080,
                "initialization": "OK",
                "status": true,
                "width (input)": 1920,
                "width (output)": 1920
            },
            "HD2K@15": {
                "fps (input)": 15,
                "fps (output)": 15,
                "height (input)": 1242,
                "height (output)": 1242,
                "initialization": "OK",
                "status": true,
                "width (input)": 2208,
                "width (output)": 2208
            },
            "HD720@15": {
                "fps (input)": 15,
                "fps (output)": 15,
                "height (input)": 720,
                "height (output)": 720,
                "initialization": "OK",
                "status": true,
                "width (input)": 1280,
                "width (output)": 1280
[... 209 lines cut ...]

</details>

### Steps to Reproduce

Since its a custom Image its hard to give clear steps to reproduce. However i linked the `image features` in the Environment section.
Regarding the docker image itself: I used the recording example from `stereolabs/zed:3.6-py-devel-jetson-jp4.6`

### Expected Result

`.svo` file in exspected location

### Actual Result

 `Segmentation fault (core dumped)` 

### ZED Camera model

ZED2

### Environment

```shell
Yocto Image based on L4T release r32.6.1 (Jetpack 4.6) with following image features: `nvidia-docker cuda-toolkit cuda-driver cuda-target-environment tensorrt-core tensorrt-plugins-prebuilt python3-tensorrt v4l-utils libv4l` Is there something Im missing out?
```

Myzhar, 2022-02-01:
Hi @timongentzsch 
we noted your bug report and we are going to analyze it as soon as possible.

adujardin, 2022-02-01:
I just check which libs were actually used by a simple program recording an SVO file on a Xavier NX JP4.6.1, here is the result:

```
/lib/aarch64-linux-gnu/libuuid.so.1.3.0
/lib/aarch64-linux-gnu/libblkid.so.1.1.0
/lib/aarch64-linux-gnu/libmount.so.1.1.0
/lib/aarch64-linux-gnu/libmount.so.1.1.0
/lib/aarch64-linux-gnu/libresolv-2.27.so
/lib/aarch64-linux-gnu/libselinux.so.1
/usr/lib/aarch64-linux-gnu/libgio-2.0.so.0.5600.4
/usr/lib/aarch64-linux-gnu/libgmodule-2.0.so.0.5600.4
/lib/aarch64-linux-gnu/libpcre.so.3.13.3
/usr/lib/aarch64-linux-gnu/tegra/libnvdla_runtime.so
/usr/lib/aarch64-linux-gnu/libglib-2.0.so.0.5600.4
/usr/lib/aarch64-linux-gnu/libgobject-2.0.so.0.5600.4
/usr/lib/aarch64-linux-gnu/tegra/libnvmmlite_image.so
/usr/lib/aarch64-linux-gnu/tegra/libnvparser.so
/usr/lib/aarch64-linux-gnu/tegra/libnvtnr.so
/usr/lib/aarch64-linux-gnu/tegra/libnvddk_2d_v2.so
/usr/lib/aarch64-linux-gnu/tegra/libnvmedia.so
/usr/lib/aarch64-linux-gnu/tegra/libnvavp.so
/usr/lib/aarch64-linux-gnu/tegra/libnvmmlite_utils.so
/usr/lib/aarch64-linux-gnu/tegra/libnvmmlite.so
/usr/lib/aarch64-linux-gnu/tegra/libnvmm_utils.so
/usr/lib/aarch64-linux-gnu/tegra/libnvbufsurface.so.1.0.0
/usr/lib/aarch64-linux-gnu/tegra/libnvtvmr.so
/usr/lib/aarch64-linux-gnu/tegra/libnvmmlite_video.so
/usr/lib/aarch64-linux-gnu/tegra/libtegrav4l2.so
/usr/lib/aarch64-linux-gnu/tegra/libv4l2_nvvideocodec.so
/usr/lib/aarch64-linux-gnu/gconv/UTF-16.so
/usr/lib/aarch64-linux-gnu/tegra/libnvidia-eglcore.so.32.6.1
/usr/lib/aarch64-linux-gnu/libffi.so.6.0.4
/usr/lib/aarch64-linux-gnu/libwayland-server.so.0.1.0
/usr/lib/aarch64-linux-gnu/libwayland-client.so.0.3.0
/usr/lib/aarch64-linux-gnu/tegra/libnvidia-egl-wayland.so
/usr/lib/aarch64-linux-gnu/libdrm.so.2.4.0
/usr/lib/aarch64-linux-gnu/tegra/libnvimp.so
/usr/lib/aarch64-linux-gnu/tegra/libnvdc.so
/usr/lib/aarch64-linux-gnu/tegra/libnvll.so
/usr/lib/aarch64-linux-gnu/tegra/libdrm.so.2
/usr/lib/aarch64-linux-gnu/tegra/libnvidia-rmapi-tegra.so.32.6.1
/usr/lib/aarch64-linux-gnu/tegra/libnvidia-glsi.so.32.6.1
/usr/lib/aarch64-linux-gnu/tegra-egl/libEGL_nvidia.so.0
/lib/aarch64-linux-gnu/libbsd.so.0.8.7
/usr/lib/aarch64-linux-gnu/libGLdispatch.so.0.0.0
/usr/lib/aarch64-linux-gnu/libXdmcp.so.6.0.0
/usr/lib/aarch64-linux-gnu/libXau.so.6.0.0
/usr/lib/aarch64-linux-gnu/tegra/libnvos.so
/usr/lib/aarch64-linux-gnu/tegra/libnvbuf_fdmap.so.1.0.0
/usr/lib/aarch64-linux-gnu/tegra/libnvddk_vic.so
/usr/lib/aarch64-linux-gnu/libEGL.so.1.0.0
/usr/lib/aarch64-linux-gnu/libxcb.so.1.1.0
/usr/lib/aarch64-linux-gnu/tegra/libnvv4lconvert.so
/lib/aarch64-linux-gnu/libudev.so.1.6.9
/usr/lib/aarch64-linux-gnu/tegra/libnvidia-fatbinaryloader.so.440.18
/usr/lib/aarch64-linux-gnu/tegra/libnvrm_graphics.so
/usr/lib/aarch64-linux-gnu/tegra/libnvrm.so
/usr/lib/aarch64-linux-gnu/tegra/libnvrm_gpu.so
/usr/lib/aarch64-linux-gnu/libgomp.so.1.0.0
/lib/aarch64-linux-gnu/libm-2.27.so
/usr/lib/aarch64-linux-gnu/tegra/libnvbuf_utils.so.1.0.0
/usr/lib/aarch64-linux-gnu/libX11.so.6.3.0
/usr/lib/aarch64-linux-gnu/tegra/libnvv4l2.so
/lib/aarch64-linux-gnu/libusb-1.0.so.0.1.0
/usr/lib/aarch64-linux-gnu/libturbojpeg.so.0.1.0
/usr/lib/aarch64-linux-gnu/libjpeg.so.8.1.2
/lib/aarch64-linux-gnu/libz.so.1.2.11
/usr/lib/aarch64-linux-gnu/libpng16.so.16.34.0
/usr/lib/aarch64-linux-gnu/tegra/libcuda.so.1.1
/lib/aarch64-linux-gnu/librt-2.27.so
/lib/aarch64-linux-gnu/libdl-2.27.so
/lib/aarch64-linux-gnu/libpthread-2.27.so
/lib/aarch64-linux-gnu/libc-2.27.so
/lib/aarch64-linux-gnu/libgcc_s.so.1
/usr/lib/aarch64-linux-gnu/libstdc++.so.6.0.25
/usr/local/zed/lib/libsl_zed.so
/lib/aarch64-linux-gnu/ld-2.27.so
```

I got the list by reading the `/proc/<PID>/maps` file, so it should also include the dynamically loaded ones. If they're all at the same location, I guess it should work

timongentzsch, 2022-02-02:
thanks @adujardin!
i solved it by manually passing the libv4l library in the docker via `-v /usr/lib/libv4l:/usr/lib/aarch64-linux-gnu/libv4l` for now. (Probably makes sense to edit the `container-runtime-csv` in the `libv4l` layer on the yocto side)

closed 2022-02-02 by timongentzsch