stereolabs/zed-docker #41: Zed2i in docker without superuser

GiteZz, 2022-02-03:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I am trying to get the ZED2i to work in a docker container without running as superuser. Several issues have come up that I could solve but now finally when the program is finally able to run as a normal user it no longer wants to connect to the camera. 

This is the dockerfile that I'm using to get it to even compile:

```
FROM company_name.azurecr.io/ros_base:cuda

USER root

ENV NVIDIA_DRIVER_CAPABILITIES \
    ${NVIDIA_DRIVER_CAPABILITIES:+$NVIDIA_DRIVER_CAPABILITIES,}compute,video,utility

RUN echo "Europe/Paris" > /etc/localtime ; echo "CUDA Version 11.4.1" > /usr/local/cuda/version.txt

# Setup the ZED SDK
RUN apt-get update -y && apt-get install --no-install-recommends lsb-release wget less udev sudo  build-essential cmake -y 

COPY ./ZED_SDK.run ./ZED_SDK.run
RUN chmod +x ./ZED_SDK.run
RUN ./ZED_SDK.run  -- silent skip_tools
RUN ln -s /lib/x86_64-linux-gnu/libusb-1.0.so.0 /usr/lib/x86_64-linux-gnu/libusb-1.0.so

RUN apt-get update -y && apt-get install libpng-dev libgomp1 -y 
RUN wget download.stereolabs.com/zedsdk/pyzed -O /usr/local/zed/get_python_api.py 

RUN python3 /usr/local/zed/get_python_api.py
RUN python3 -m pip install numpy opencv-python *.whl 

RUN apt-get install -y ros-galactic-diagnostic-updater
RUN apt-get install -y ros-galactic-xacro

ENV NVIDIA_VISIBLE_DEVICES \
    ${NVIDIA_VISIBLE_DEVICES:-all}

RUN chown -R company_name:company_name /usr/local/lib/python3.8/dist-packages
RUN chown -R company_name:company_name /usr/local/zed
RUN chown company_name:company_name /usr/lib/x86_64-linux-gnu/libusb-1.0.so
RUN chown company_name:company_name /lib/x86_64-linux-gnu/libusb-1.0.so.0 

ENV LD_LIBRARY_PATH $LD_LIBRARY_PATH:/usr/local/zed/lib/

USER company_name

```

The `company_name.azurecr.io/ros_base:cuda` is a docker image build on top of `nvidia/cuda:11.4.1-cudnn8-devel-ubuntu20.04` with ROS galactic added. When not running as the `company_name` user everything seemed to work. However when run with this user we get this output:

```
[zed_wrapper-2] 1643885217.365956459 [zed2i.zed_node] [INFO] ********************************
[zed_wrapper-2] 1643885217.366047245 [zed2i.zed_node] [INFO]       ZED Camera Component 
[zed_wrapper-2] 1643885217.366058650 [zed2i.zed_node] [INFO] ********************************
[zed_wrapper-2] 1643885217.366065898 [zed2i.zed_node] [INFO]  * namespace: /zed2i
[zed_wrapper-2] 1643885217.366072904 [zed2i.zed_node] [INFO]  * node name: zed_node
[zed_wrapper-2] 1643885217.366079143 [zed2i.zed_node] [INFO] ********************************
[zed_wrapper-2] 1643885217.366092624 [zed2i.zed_node] [INFO] *** DEBUG parameters ***
[zed_wrapper-2] 1643885217.366121072 [zed2i.zed_node] [INFO]  * Debug mode: FALSE
[zed_wrapper-2] 1643885217.366135747 [zed2i.zed_node] [INFO] *** GENERAL parameters ***
[zed_wrapper-2] 1643885217.366171277 [zed2i.zed_node] [INFO]  * Camera model: zed2i - ZED 2i
[zed_wrapper-2] 1643885217.366208824 [zed2i.zed_node] [INFO]  * SDK Verbose: 1
[zed_wrapper-2] 1643885217.366246731 [zed2i.zed_node] [INFO]  * SVO: 
[zed_wrapper-2] 1643885217.366282283 [zed2i.zed_node] [INFO]  * SVO Loop: TRUE
[zed_wrapper-2] 1643885217.366321488 [zed2i.zed_node] [INFO]  * SVO Realtime: FALSE
[zed_wrapper-2] 1643885217.366360800 [zed2i.zed_node] [INFO]  * Camera name: zed2i
[... 183 lines cut ...]
1. Build and run docker image
2. Source ROS
3. Launch the provided ROS launch file `ros2 launch zed_wrapper zed2i.launch.py`
...

### Expected Result

Connection to the camera

### Actual Result

Traceback listed above.

### ZED Camera model

ZED2i

### Environment

```shell
OS: nvidia/cuda:11.4.1-cudnn8-devel-ubuntu20.04 docker on Ubuntu 20.04 host OS
CUDA: 11.4.1
ZED_SDK: 3.6.4
GPU: RTX3000
```

adujardin, 2022-02-03:
The typical issue is the missing dev rules on the host but you seem to have fixed this already.

To verify that the camera is correctly seen in the container could your post the output of `lsusb` in the container and on the host? 

To clarify, running the same docker image but as root is working as expected?
Could you also try to run a simple C++ tutorial to see if you have the same behavior, just in case?

GiteZz, 2022-02-03:
Thanks for the reply!

This is the output of `lsusb` in the **container**:

```
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 002: ID 174c:3074 ASMedia Technology Inc. ASM1074 SuperSpeed hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 003: ID 0572:180a Conexant Systems (Rockwell), Inc. HP Dock Audio
Bus 005 Device 002: ID 174c:2074 ASMedia Technology Inc. ASM1074 High-Speed hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 004: ID 8564:1000 Transcend Information, Inc. JetFlash
Bus 002 Device 003: ID 2109:0813 VIA Labs, Inc. USB3.0 Hub
Bus 002 Device 072: ID 2b03:f880 Technologies, Inc. ZED 2i
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 06cb:00b7 Synaptics, Inc. 
Bus 001 Device 005: ID 0408:5390 Quanta Computer, Inc. HP Full-HD Camera
Bus 001 Device 010: ID 046d:c077 Logitech, Inc. M105 Optical Mouse
Bus 001 Device 008: ID 256f:c635  
Bus 001 Device 006: ID 413c:2113 Dell Computer Corp. 
Bus 001 Device 003: ID 2109:2813 VIA Labs, Inc. USB2.0 Hub
Bus 001 Device 009: ID 8087:0029 Intel Corp. 
Bus 001 Device 013: ID 2b03:f881  
Bus 001 Device 002: ID 0424:2512 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

It seems that the ZED2i is visible inside the containers. 
This is my output on the **host**:

```
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 002: ID 174c:3074 ASMedia Technology Inc. ASM1074 SuperSpeed hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 003: ID 0572:180a Conexant Systems (Rockwell), Inc. HP Dock Audio
Bus 005 Device 002: ID 174c:2074 ASMedia Technology Inc. ASM1074 High-Speed hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 004: ID 8564:1000 Transcend Information, Inc. JetFlash
Bus 002 Device 003: ID 2109:0813 VIA Labs, Inc. USB3.0 Hub
Bus 002 Device 006: ID 2b03:f880 Technologies, Inc. ZED 2i
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 06cb:00b7 Synaptics, Inc. 
Bus 001 Device 005: ID 0408:5390 Quanta Computer, Inc. HP Full-HD Camera
Bus 001 Device 010: ID 046d:c077 Logitech, Inc. M105 Optical Mouse
Bus 001 Device 008: ID 256f:c635  
Bus 001 Device 006: ID 413c:2113 Dell Computer Corp. 
Bus 001 Device 003: ID 2109:2813 VIA Labs, Inc. USB2.0 Hub
Bus 001 Device 009: ID 8087:0029 Intel Corp. 
Bus 001 Device 013: ID 2b03:f881  
Bus 001 Device 002: ID 0424:2512 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

I have also tested the c++ tutorials and this is my output when running the first tutorial:

```
[ZED][Init][Info] Camera opening timeout reached
Error CAMERA FAILED TO SETUP, exit program.
```

adujardin, 2022-02-03:
Ok, thanks. So both the Camera (`2b03:f880`) and IMU (`2b03:f881`) are correctly detected.

`CAMERA FAILED TO SETUP` usually means there's an issue with the video device (/dev/videoX). Could you try adding the user to the `video` group?

GiteZz, 2022-02-03:
The docker user was indeed not in the video group, however after adding the user to the group and verifying with `cat /etc/group | grep video` I got the same errors as above.

adujardin, 2022-02-03:
Ok, and If you use the default zed docker image such as `stereolabs/zed:3.6-devel-cuda11.4-ubuntu20.04` do you have the same issue as non-root user?

Could you also try to run something that would open the ZED as a webcam in your docker image? If you don't have a GUI, maybe a simple Opencv VideoCapture program or something to check if the camera can be opened, including taking an image from the camera as a file to make sure.

Bortronx, 2022-02-03:
> Ok, and If you use the default zed docker image such as `stereolabs/zed:3.6-devel-cuda11.4-ubuntu20.04` do you have the same issue as non-root user?
>
> Could you also try to run something that would [...]

I do have it displaying GUI so I installed cheese and had no video input there either. I tried using `--device=/dev/video0` and I also had no video show up.

adujardin, 2022-02-04:
@GiteZz It just occurs to me but you may need to call `newgrp video` for the group change to take effect, depending on how you added the user to the group.

obraun-sl, 2022-02-08:
Hi,
A recent update in V4L driver could cause such issue. 
Make sure you use the ZED SDK 3.6.5 that supports it , or in case of Docker, make sure you force-pull the last version of 3.6

closed 2022-04-27 by github-actions[bot]

zymouse, 2022-07-20:
Hello, encountered a problem with.
Description 1: lsusb not detected - camera [2b03:f880]
Description 2: During use or the next boot, this condition will appear lsusb can not be the camera [as follows]
speculation: whether it is a hardware problem
Used four monocular cameras and a zed, I have six usb3.0 IPCs
Thank you very much for your help!
[image]

zymouse, 2022-07-20:
Hello, encountered a problem with.
Description 1: lsusb not detected - camera [2b03:f880]
Description 2: During use or the next boot, this condition will appear lsusb can not be the camera [as follows]
speculation: whether it is a hardware problem
Used four monocular cameras and a zed, my IPC has six usb3.0
Thank you very much for your help!
[image]