stereolabs/zed-docker #42: Zed Explorer not detecting USB Camera on Ubuntu 20.04

Bortronx, 2022-02-03:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

After following the steps on the [Install Guide](https://www.stereolabs.com/docs/docker/install-guide-linux/) 
 when launching ZED Explorer with the camera connected there is no camera feed input. The camera works fine and is detected by Ubuntu. The docker container is in `--privileged` mode.

I am using the following command to run it:

`sudo docker run -it --gpus all --privileged -e DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix stereolabs/zed:3.6-gl-devel-cuda11.4-ubuntu20.04`

Zed Explorer opens but black. No video feed.

Nvidia cuda docker is installed and running. Everything works but no camera is detected.

### Steps to Reproduce

1. Follow Installation guide  [Install Guide](https://www.stereolabs.com/docs/docker/install-guide-linux/) 
2. Open ZED_Explorer using `/usr/local/zed/tools/ZED_Explorer`
3. Wait for Opening message
...

### Expected Result

Camera Feed on ZED_Explorer

### Actual Result

No Camera Feed on ZED_Explorer window

### ZED Camera model

ZED Mini

### Environment

```shell
**OS:** Ubuntu 20.04 
**CPU:** Intel CORE i7-9750H 9th Gen x86_64
**GPU:** Nvidia GP107GLM Quadro P1000 Mobile
**ZED SDK version:** v3.6.0
**Other:** USB 3.0
```

### Anything else?

lsusb returns:
Bus 002 Device 004: ID 2b03:f682 Technologies, Inc. ZED-M

obraun-sl, 2022-02-04:
Hi,

Just to make sure, Does ZED Explorer works when not in docker ? To make sure the problem is not on the camera itself. 

Normally lsusb in container should give you 2 devices : 
2b03:F682 --> ZED-M video controller
2b03:F681 --> ZED-M sensor controller

Can you check?

Bortronx, 2022-02-04:
Hello!

thank you for the help. I get both in lsusb the second one shows as following:
`Bus 001 Device 007: ID 2b03:F681STEREOLABS ZED-M Hid Device`

ZED Explorer works in Windows on a different machine. Also I can see the camera feed fine using cheese outside of my zed-docker container. The problem is when using docker specifically.

bblumberg, 2022-02-05:
I have run into the same thing and I suspect it may be due to a very recent upgrade because my camera was working fine yesterday and after doing what I thought was going to be a routine upgrade it stopped working (camera not detected). The camera does work on my jetson, so I know it is not the camera and it is not the docker setup (doesn't work outside of docker either).

obraun-sl, 2022-02-06:
Hi,

It seems the latest Ubuntu update upgraded the v4l2 driver and it changed the way it handles a parameter, hence breaking the connection with the ZED cameras.
The problem has been identified and we will release the patch v3.6.5 as soon as the QA tests are completed.
The fix will be available to download in the next few day.

bblumberg, 2022-02-06:
Thanks for the quick turn around! And I’m hoping the patch will be applied
both to the sdk and the docker version.

Thanks again for looking into this so quickly.

bb

adujardin, 2022-02-07:
Just to let you know that the docker images are also up to date with 3.6.5 now. You may have to re-pull manually the 3.6 tags

bblumberg, 2022-02-08:
Everything works for me, so feel free to close this. Thanks again for your quick reponse!

closed 2022-04-27 by github-actions[bot]