stereolabs/zed-docker #46: xhost:  unable to open display ""

nstepanovs412, 2023-01-05:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hello, 
This is the first run of ZED 2 Stereolabs on NVIDIA Jetson Nano. 
When running
`sudo docker run --gpus all -it --privileged stereolabs/zed:3.0-devel-jetson-jp4.2`
The following errors appear
`WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested
root@ba9e02dde91c:/usr/local/zed# xhost +si:localuser:root
xhost:  unable to open display ""`

Please assist

### Steps to Reproduce

1. Run `sudo docker run --gpus all -it --privileged stereolabs/zed:3.0-devel-jetson-jp4.2
`
2. Get error `WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested
root@ba9e02dde91c:/usr/local/zed# xhost +si:localuser:root
xhost:  unable to open display ""`

### Expected Result

Open display 

### Actual Result

`WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested
root@ba9e02dde91c:/usr/local/zed# xhost +si:localuser:root
xhost:  unable to open display ""
`

### ZED Camera model

ZED2

### Environment

```shell
OS: Nvidia Jetson Ubuntu
Version: Nvidia Jetson Nano
Camera: Zed 2
```

adujardin, 2023-01-06:
Hi,
The command `xhost +si:localuser:root` should run on the host, not in the container. 
Also are you sure you're using the Jetpack 4.2 on your Jetson ? It's now quite old and the version must match exactly to work

nstepanovs412, 2023-01-06:
Hello,
Thanks for the answer. 
I will try to run `xhost +si:localuser:root` as noted. 

About Jetpack:
I just flashed the most recent version of NVIDIA OS following the "getting started guide" https://developer.nvidia.com/embedded/learn/get-started-jetson-nano-devkit

Thanks.

adujardin, 2023-01-06:
Ok, so you should have JP4.6.1. You should use `stereolabs/zed:3.8-devel-jetson-jp4.6.1` or `stereolabs/zed:3.7-devel-jetson-jp4.6.1`

nstepanovs412, 2023-01-06:
Thanks, the issue has been resolved.

closed 2023-01-06 by nstepanovs412