stereolabs/zed-docker #49: ZED_Depth_Viewer crashes on Jetson AGX Orin

mandulaj, 2023-06-22:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

We are trying to get the camera working on a Nvidia Jetson AGX Orin but the the ZED_Depth_Viewer crashes with the following Segmentation fault:

```
nvbuf_utils: Could not get EGL display connection
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[ZED Depth Viewer] Failed to create directory to save images: /root/Documents/ZED
QGLShader::compile(Vertex): 0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.40, 1.00 ES, and 3.00 ES

[ZED Depth Viewer] Failed to compile Vertex shader GLSL 
QGLShader::compile(Fragment): 0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.40, 1.00 ES, and 3.00 ES

[ZED Depth Viewer] Failed to compile fragment shader GLSL 
QGLShader::link: "error: linking with uncompiled/unspecialized shadererror: linking with uncompiled/unspecialized shader"
[ZED Depth Viewer] Failed to link shaders to program
QGLShader::link: "error: linking with uncompiled/unspecialized shadererror: linking with uncompiled/unspecialized shader"
[ZED Depth Viewer] Failed to bind shaders to program
QGLShaderProgram::uniformLocation( texture ): shader program is not linked
QGLShaderProgram::uniformLocation( matrix ): shader program is not linked
QGLShader::compile(Vertex): 0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.40, 1.00 ES, and 3.00 ES

[ZED Depth Viewer] Failed to compile Vertex shader GLSL 
QGLShader::compile(Fragment): 0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.40, 1.00 ES, and 3.00 ES

[ZED Depth Viewer] Failed to compile Vertex shader GLSL 
QGLShader::link: "error: linking with uncompiled/unspecialized shadererror: linking with uncompiled/unspecialized shader"
QGLShader::link: "error: linking with uncompiled/unspecialized shadererror: linking with uncompiled/unspecialized shader"
QGLShaderProgram::uniformLocation( texture ): shader program is not linked
in bool ImageHandler::initialize(sl::Mat&) : Err [999]: unknown error.
Stack trace (most recent call last):
#27   Object "/usr/local/zed/tools/ZED_Depth_Viewer", at 0x41ec07, in 
#26   Object "/usr/lib/aarch64-linux-gnu/libc.so.6", at 0xffff8405fe0f, in __libc_start_main
#25   Object "/usr/local/zed/tools/ZED_Depth_Viewer", at 0x41e187, in 
#24   Object "/usr/local/zed/tools/ZED_Depth_Viewer", at 0x43eeb3, in 
#23   Object "/usr/local/zed/tools/ZED_Depth_Viewer", at 0x43a9e3, in 
#22   Object "/usr/local/zed/tools/ZED_Depth_Viewer", at 0x44d58f, in 
#21   Object "/usr/lib/aarch64-linux-gnu/libQt5Core.so.5", at 0xffff846d03b7, in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)
#20   Object "/usr/lib/aarch64-linux-gnu/libQt5Core.so.5", at 0xffff8472d1cb, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#19   Object "/usr/lib/aarch64-linux-gnu/libglib-2.0.so.0", at 0xffff82b7bc53, in g_main_context_iteration
#18   Object "/usr/lib/aarch64-linux-gnu/libglib-2.0.so.0", at 0xffff82b7bbb3, in 
#17   Object "/usr/lib/aarch64-linux-gnu/libglib-2.0.so.0", at 0xffff82b7b943, in g_main_context_dispatch
#16   Object "/usr/lib/aarch64-linux-gnu/libQt5Core.so.5", at 0xffff8472ce73, in 
#15   Object "/usr/lib/aarch64-linux-gnu/libQt5Core.so.5", at 0xffff8472c507, in QTimerInfoList::activateTimers()
#14   Object "/usr/lib/aarch64-linux-gnu/libQt5Core.so.5", at 0xffff846d1c0b, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#13   Object "/usr/lib/aarch64-linux-gnu/libQt5Widgets.so.5", at 0xffff8508aad7, in QApplication::notify(QObject*, QEvent*)
#12   Object "/usr/lib/aarch64-linux-gnu/libQt5Widgets.so.5", at 0xffff850814ab, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#11   Object "/usr/lib/aarch64-linux-gnu/libQt5Core.so.5", at 0xffff846ff5b7, in QObject::event(QEvent*)
#10   Object "/usr/lib/aarch64-linux-gnu/libQt5Core.so.5", at 0xffff8470c557, in QTimer::timeout(QTimer::QPrivateSignal)
#9    Object "/usr/lib/aarch64-linux-gnu/libQt5Core.so.5", at 0xffff846febff, in QMetaObject::activate(QObject*, int, int, void**)
#8    Object "/usr/local/zed/tools/ZED_Depth_Viewer", at 0x41f1f3, in 
#7    Object "/usr/local/zed/tools/ZED_Depth_Viewer", at 0x43feab, in 
#6    Object "/usr/local/zed/tools/ZED_Depth_Viewer", at 0x43faab, in 
#5    Object "/usr/local/zed/tools/ZED_Depth_Viewer", at 0x43b5ef, in 
#4    Object "/usr/local/zed/tools/ZED_Depth_Viewer", at 0x4479b7, in 
#3    Object "/usr/local/zed/tools/ZED_Depth_Viewer", at 0x446893, in 
#2    Object "/usr/local/zed/tools/ZED_Depth_Viewer", at 0x4a6e27, in 
#1    Object "/usr/local/zed/tools/ZED_Depth_Viewer", at 0x4669e7, in 
#0    Object "/usr/lib/aarch64-linux-gnu/tegra/libcuda.so.1", at 0xffff89cfa1cc, in 
Segmentation fault (Address not mapped to object [0x4000400000018])
Segmentation fault (core dumped)
```

### Steps to Reproduce

1. Pull and run the Jetson tools-devel docker container: `docker run --runtime nvidia --net=host  -it --privileged -e DISPLAY -v /tmp/.X11-unix:/tmp.X11-unix -v "$HOME/.Xauthority:/root/.Xauthority" stereolabs/zed:4.0-tools-devel-jetson-jp5.1.0` in
2. Inside the docker container run `/usr/local/zed/tools/ZED_Depth_Viewer`
3. ... Crash happens
...

### Expected Result

The application starts, connects to the camera and shows the 3D depth map.

### Actual Result

The application starts, connects to the camera and crashes with the Segmentation fault.

### ZED Camera model

ZED2

### Environment

```shell
NVIDIA Jetson AGX Orin
 L4T 35.2.1 [ JetPack UNKNOWN ]
   Ubuntu 20.04.6 LTS
   Kernel Version: 5.10.104-tegra
 CUDA 11.4.315
   CUDA Architecture: 8.7
 OpenCV version: 4.5.4
   OpenCV Cuda: NO
 CUDNN: 8.6.0.166
 TensorRT: 8.5.2.2
 Vision Works: NOT_INSTALLED
 VPI: 2.2.4
 Vulcan:
```

prakharg01, 2024-05-08:
Issue:
Encountering crashes when running tools from the ZED SDK within the Nvidia Isaac ROS Docker environment. Suspected mismatch between ZED SDK versions in Dockerfile and required version for ZED Box Orin NX (16GB).

Description:
Attempting to set up ZEDX with Nvidia Isaac ROS Docker following the provided documentation (https://nvidia-isaac-ros.github.io/getting_started/hardware_setup/sensors/zed_setup.html). ZEDX driver is functional, but when executing tools from the SDK, consistent crashes occur. Suspected cause may be a discrepancy between the ZED SDK version specified in the Dockerfile (found at https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/blob/main/docker/scripts/install-zed-aarch64.sh) and the required version for my hardware (ZED Box Orin NX (16GB)).

Action Needed:

Confirm compatibility between ZED SDK versions specified in Dockerfile and required version for ZED Box Orin NX. If mismatch is confirmed, update Dockerfile to include the required ZED SDK version (ZED SDK for JetPack 5.1.1 (L4T 35.3), version 4.1).

Any help or comments will be appreciated.

prakharg01, 2024-05-08:
Update:

Successfully updated the Dockerfile with the correct SDK version, resolving the previous issue. Added the line wget -q --no-check-certificate -O ZED_SDK_Linux.run https://download.stereolabs.com/zedsdk/4.1/l4t35.3/jetsons to fetch the required SDK version.

However, a warning persists:

[ZED-X][Warning] Failed to connect to zed_x_daemon.
[ZED-X][Warning] Failed to verify the match between the L4T version of the host and the container. If you are indeed in a container, be sure to run the same L4T version on both.

When checking cat /etc/nv_tegra_release within the container, it returns "not found," while in the host, it returns:

# R35 (release), REVISION: 3.1, GCID: 32827747, BOARD: t186ref, EABI: aarch64, DATE: Sun Mar 19 15:19:21 UTC 2023

Despite the successful operation of the tools, the warning persists. Is this something that needs to be addressed?