stereolabs/zed-docker #51: Provide a 4.0-gl-devel-cuda12.1-ubuntu22.04

zaporter, 2023-09-05:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

I'd love to have a `gl` version of the docker container for ubuntu 22.04.

### Use-Case

For some reason, the ZED-mini camera is not detected inside of docker on 20.04
`lsusb` on my host:
```
Bus 001 Device 013: ID 2b03:f681 STEREOLABS ZED-M HID Interface
```

`lsusb` in `stereolabs/zed:4.0-gl-devel-cuda11.4-ubuntu20.04`:
```
Bus 001 Device 008: ID 1e91:de41
```
(and it ZED_Explorer never finds the camera)

`lsusb` in `stereolabs/4.0-devel-cuda12.1-ubuntu22.04`:
```
Bus 001 Device 013: ID 2b03:f681 STEREOLABS ZED-M HID Interface
```

### Anything else?

My host:
```
Linux host 6.1.45 #1-NixOS SMP PREEMPT_DYNAMIC Fri Aug 11 10:08:27 UTC 2023 x86_64 GNU/Linux
```

I thought about making this a bug report, but it seems like other people have had issues with ubuntu 20.04 and the advice was just to upgrade. Sorry if I used the wrong tag!

Thanks!

adujardin, 2023-09-06:
Hi, 
I'd also love that, but it's currently blocked by Nvidia support of cudagl base images. It's no longer receiving updates and doesn't support Ubuntu 22. More info: https://gitlab.com/nvidia/container-images/cuda/-/issues/147
We would like to avoid having to manually create the base image and use the official ones.

zaporter, 2023-09-10:
> We would like to avoid having to manually create the base image and use the official ones.

Ahh -- I see, that makes sense. Well, thank you for your time! (loving the camera!)